rsmp-nordic/rsmp_simulator #1: Status messages always return null

emiltin, 2019-06-14:
I'm having trouble getting statuses to work with the site simulator. Is it implemented?

After connection is established, I send a StatusRequest:
`<--  41d3  Sent StatusRequest {"mType":"rSMsg","type":"StatusRequest","ntsOId":"","xNId":"","cId":"AA+BBCCC=DDDEE002","sS":[{"cCI":"S001","n":"number"}],"mId":"41d3911c-e509-48e7-8ce8-70c2c8a21d55"}
`

And get back a StatusResponse, but sCI and s is always null, and q is undefined:
`-->  41f3  Received StatusResponse {"mType":"rSMsg","type":"StatusResponse","mId":"41f38171-f3f9-434e-a068-9340d5380a11","ntsOId":"","xNId":"","cId":"AA+BBCCC=DDDEE002","sTs":"2019-06-14T07:39:04.358Z","sS":[{"sCI":null,"n":"number","s":null,"q":"undefined"}]}
`

This seems to happen, no matter what component id, status code id or name I use. I see a similar behaviour when trying to subscribe to statuses.

I'm using the default site settings in the simulator, which is a VMS sign:
[image]

tomasrook, 2020-03-10:
Ok, it is because you use CommandCodeId 'cCI' instead of 'sCI'. I have updated the code to evaluate the sS's first before responding individually and it now sends a MessageNotAck with reason:
_StatusCode Id (sCI) in StatusRequest is missing_

otterdahl, 2020-03-27:
Fixed in 1.0.1.5

closed 2020-03-27 by otterdahl