rsmp-nordic/rsmp_simulator #4: Invalid messages during connection causes stall

emiltin, 2019-06-14:
When invalid messages are send during connection sequence, the simulator stalls and does not send anything more, nor does it disconnect.

tomasrook, 2020-03-10:
I have made some changes so it by default will close the connection when initial sequence is out of order (if the simulator detects other packets if negotioation is not finished). I have added some settings to test this:

_Drop random bytes in negotiation packets (10% dropped)
Close connection if negotiation is out of sequence_

However, the simulator does not resend version packets if not getting any ack/nack - this is by design to make it easier to follow the intial packets in the debugger without interfering retries. Nothing more happens with connection left open which will be a 'stall'. But I can change the behaviour and make it optional to turn off retransmission which probably will be a correct behaviour.

otterdahl, 2020-03-27:
Fixed in 1.0.1.5

closed 2020-03-27 by otterdahl

emiltin, 2020-03-28:
Will 1.0.15 resend version packets if not getting any ack/nack - as is the correct behaviour according to spec?