rsmp-nordic/rsmp_simulator #6: Only auto-scroll log window if already at bottom

emiltin, 2019-06-14:
System log should not auto-scroll to bottom when new message are received, unless it's already at the bottom.

The current behaviour means it's very hard to look at earlier messages, since every new messages causes the window to auto-scroll.

tomasrook, 2019-06-27:
Ok, good point.
I have updated the behaviour to auto scroll to the last item only if no item is selected or the last item is selected. If you select some other item in the list, it will remain selected and the list is filled without scrolling. Seemed to be the logical solution since there is not much place to add a 'Always show last row' checkbox.
Is fixed in release 1.0.1.4 B4

closed 2019-06-27 by tomasrook

emiltin, 2019-07-01:
Great