rsmp-nordic/rsmp_simulator #7: Installers cannot run silently, incorrect check for .NET

emiltin, 2019-07-03:
To get the simulator approved for installation on PC's at our organisation, the software must support silent intallations. It seems there are some minor problems preventing this:

RSMPGS2_1_0_1_1_Setup.exe has no parameters for running silently.

RSMPGS1_1_0_1_2_Setup.exe has parameters for running silently, but there's an incorrect check causing the install to complain about .NET missing, even though it IS installled: "Microsoft .NET Framework 3.5 SP1 does not seem to be installed, RSMPGS1 amy have problems running without it. Do you really want to continue setup?"

tomasrook, 2020-03-09:
The RSMPGS2_1_0_1_1_Setup.exe was probably built with some older version of InnoSetup which would explain why it did not support silent install (it was built in 2012)
The .NET installation check is made by simply using this script:
_DirExists(GetWinDir + '\Microsoft.NET\Framework\v3.5\Microsoft .NET Framework 3.5 SP1');_
This was fine in the early days but does not really work anymore.

Next release (2) is updated to VS2015 and .NET Framework 4.6.1.

I suggest we simply remove the .NET install check and update the documentation to be more clear what framework is needed.

otterdahl, 2020-03-27:
Fixed in 1.0.1.5

closed 2020-03-27 by otterdahl