rsmp-nordic/rsmp_simulator #8: Expects wrongly cased StatusUnsubscribe messages type

emiltin, 2019-07-09:
According to the RSMP core specifikation, status unsubscribe message must use the type "StatusUnsubscribe". But when sending message with this type to the simulator, it complains that it does not match "StatusUnSubscribe".

```
<--  Sent StatusUnsubscribe {"mType":"rSMsg","type":"StatusUnsubscribe","ntsOId":"","xNId":"","cId":"AA+BBCCC=DDDEE002","sS":[{"sCI":"S001","n":"number"}],"mId":"f0dc6ba4-debe-49d5-8ba4-8b6befb96797"}

-->  Received MessageNotAck for StatusUnsubscribe f0dc {"mType":"rSMsg","type":"MessageNotAck","oMId":"f0dc6ba4-debe-49d5-8ba4-8b6befb96797","rea":"Invalid property, \u0027StatusUnsubscribe\u0027 does not match \u0027StatusUnSubscribe\u0027"}
```

tomasrook, 2020-03-09:
According to "Versionshistorik RSMP-Kommunikationsprotokoll vägsidesutrustning.docx" it was a spelling error corrected in 3.1.2, however the text says it was corrected from "StatusUnsubscribe" to "StatusUnSubscribe". I assume it is the opposite to reflect rsmp-spec-3.1.4. I have changed the simulator to validate using "StatusUnsubscribe". It will be in the next release (2).

otterdahl, 2020-03-27:
Fixed in 1.0.1.5

closed 2020-03-27 by otterdahl