rsmp-nordic/rsmp_simulator #12: add REAME file

emiltin, 2020-03-30:
Add a readme file with:

- about the project
- how to install
- where to find documentation
- where to report issues and suggest improvements

otterdahl, 2020-05-04:
Readme added with 233bb91eb7f883249d737c562ddb86c0693c2737

closed 2020-05-04 by otterdahl

emiltin, 2020-05-04:
I suggest mentioning in the readme that it's a program that runs on Windows.

otterdahl, 2020-05-04:
Fixed.

Regarding platform support: It is possible to compile and run the RSMP simulator for Linux (and MacOS) too.The **rsmp_support** branch fixes some cross-platfrom issues. Not everything works 100% and it is a bit experimental. To compile just download  [monodevelop](https://www.monodevelop.com/) (or similar) and build it.

emiltin, 2020-05-04:
ok didn't know that