rsmp-nordic/rsmp_simulator #18: RSMPGS1 dependent on locale for correct format of timestamp

otterdahl, 2021-05-03:
When using Finnish locale the timestamp produced by RSMPGS1 are invalid.

```
Info   14.30.39.653 Got message, Type: AggregatedStatus, mId: 579d14e2-2139-4079-bbdd-deee5a3355c6
Error  14.30.39.660 Could not deserialize timestamp '2021-05-03T11.30.39.521Z', error: String was not recognized as a valid DateTime.
```

Changing the locale to Swedish makes the error disappear.

emiltin, 2021-05-04:
this is an issue with the simulator, not the spec?

otterdahl, 2021-05-04:
Correct

emiltin, 2021-05-04:
ok. but maybe we should check if the validator or the rsmp gem could have similar issues.

otterdahl, 2021-09-27:
Partially fixed by PR #22. Some extra fixes needed

otterdahl, 2021-09-27:
No extra fixes needed. Closing issue

closed 2021-09-27 by otterdahl