rsmp-nordic/rsmp_simulator #19: Simulator 1.0.1.6 doesn't handle timestamps properly.

pjbouwens, 2021-08-24:
When a backlog of events is sent to the RSMPGS2 the events are handled as they arrive instead of inspecting the timestamps reported. 
 
The following A0003 alarms are sent to the RSMPGS2 in the following order. The first message sent is the one with the newest timestamp. 
 
LOG_DEBUG... 2021/08/17 15:35:34: [ >> [10.164.68.200:12111] {"mType" : "rSMsg","type" : "Alarm","mId" : "bb49286a-4bc7-4eb1-92a7-34fed98c3862","ntsOId" : "KK+AG1404=001TC000","xNId" : "00001","cId" : "KK+AG1404=001TC000","aCId" : "A0003","xACId" : "","xNACId" : "110263","aSp" : "issue","ack" : "notAcknowledged","aS" : "**active**","sS" : "notSuspended","aTs" : "**2021-08-17T13:35:32.000Z**","cat" : "D","pri" : "2","rvs" : []}]
LOG_DEBUG... 2021/08/17 15:35:34: [ >> [10.164.68.200:12111] {"mType" : "rSMsg","type" : "Alarm","mId" : "98b6a91c-d34f-4a35-8577-6b40ce1be649","ntsOId" : "KK+AG1404=001TC000","xNId" : "00001","cId" : "KK+AG1404=001TC000","aCId" : "A0003","xACId" : "","xNACId" : "","aSp" : "issue","ack" : "notAcknowledged","aS" : "**inactive**","sS" : "notSuspended","aTs" : "**2021-08-17T13:35:28.000Z**","cat" : "D","pri" : "2","rvs" : []}]

Expected is that the Alarm status of the RSMPGS2 is shown as “inactive”, but it ends up being shown as “active” on the GUI. This makes it hard to use the RSMPGS2 as a validation of the functionality.

emiltin, 2021-08-24:
Thank you for you report. Looking in the core spec, it's not clear to me how the receiver should handle messages with timestamps that are not sequential.

But the core specification states in section 4.3.4 that "The communication buffer is stored and sent using the FIFO principle".  See https://rsmp-nordic.github.io/rsmp_core/applicability/transport_of_data.html?highlight=buffer#communication-disruption

This seems to imply that you should not send messages with timestamps that are not in sequential order, and that it might even be an error to do so.

emiltin, 2021-08-24:
See also these issue about continuity of timestamps:

https://github.com/rsmp-nordic/rsmp_validator/issues/84
https://github.com/rsmp-nordic/rsmp_implementations/issues/10

otterdahl, 2022-05-13:
I agree that this seems to be an issue with the simulator. The following steps are part of the handshake according to RSMP 3.1.5: :

- 8, All alarms (incl. active, inactive, suspended, unsuspended and acknowledged) are sent. (according to section [Alarm messages](https://rsmp-nordic.org/rsmp_specifications/core/3.1.5/applicability/basic_structure.html#alarm-messages)).
- 9, Buffered messages in the equipment’s outgoing communication buffer are sent, incl. alarms, aggregated status and status updates.

This would mean that during communication establishment, buffered alarm messages can be sent out-of-order. But during normal communication FIFO principle applies.

otterdahl, 2023-10-24:
Should be fixed with edf6207ff6a29c94ce3d16e04df6ea2853a722a9. Part of the upcoming 1.0.4 release

otterdahl, 2023-10-25:
Should be fixed with 1.0.4 release

closed 2023-10-25 by otterdahl