rsmp-nordic/rsmp_simulator #27: YAML: Interpret integer_list, long_list and real_list

otterdahl, 2021-09-23:
## Background

RSMP [defines the following data types](http://rsmp-nordic.org/rsmp_core/applicability/basic_structure.html?highlight=integer#return-values) for use in alarms, status and commands:
- string
- integer
- long
- real
- boolean
- base64

Each data field is either referred to as `rvs` (return values) or `arg` (arguments) in the specification. They typically use either a predefined list of allowed values or a `max` and `min` value. The max and min value was typically expressed as `[0-12]` in the `Value` column in the Excel SXL. When the YAML format was added, max and min were moved to its separate `min` and `max` fields.

Example of usage of `max` and `min` values:
```yaml
          vehicles:
            type: integer_list
            description: |
              Number of vehicles
              - Value expressed as an integer with a range of 0-65535.
              - Contains data from all detector logics. Each detector logic is separated with a comma.
              - The value is set to “-1” if no data could be measured (e.g. detector fault)
            min: -1
            max: 65535
```

Example of usage of a predefined list:
```yaml
          status:
            type: integer
            values:
              0: disabled
              1: dark mode
              2: yellow flash
```

## Data types used in the YAML file

When creating a JSON schema based on the YAML file, it's needed to know whether the data type use a predefined list of allowed values or if it uses `max` and `min` values. Therefore, the follow data types are added to the YAML file to be used when `max` and `min` values are used:
- integer_list
- long_list
- real_list

## Needed changes to the RSMP simulator

Right now, the RSMP Simulator interprets `integer_list` as `unknown`

The RSMP Simulator needs to interpret:
- `integer_list` and `integer`
- `long_list` as `long`
- `real_list` as `real`

emiltin, 2021-09-23:
> When creating a JSON schema based on the YAML file, it's needed to know whether the data type use a predefined list of allowed values or if it uses max and min values. Therefore, the follow data types are added to the YAML file to be used when max and min values are used

Not sure I follow this part. integer_list and friends are not related to value lists (enums). they are used e.g. when you have multiple intersections and need to send a value for each.

emiltin, 2021-09-23:
the rsmp gem includes a command-line option to convert an sxl from yaml to json schema. this includes the *_list types.
https://github.com/rsmp-nordic/rsmp/tree/master/lib/rsmp/convert

otterdahl, 2021-09-27:
> Not sure I follow this part. integer_list and friends are not related to value lists (enums). they are used e.g. when you have multiple intersections and need to send a value for each.

Right. I misunderstood this part

otterdahl, 2023-07-08:
The RSMP simulator doesn't support treating the comma separated values in any special way and there is no current plans change this. It is probably a better idea to use something else than comma separated lists anyway.

The YAML SXL included in the setup packages removes the "_list" suffix. This is done by the workflow. This issue can be closed.

closed 2023-07-08 by otterdahl