rsmp-nordic/rsmp_simulator #31: Suspension/acknowledgement of alarm does not affect all alarms of component

daniyalyasin93, 2023-01-30:
According to the spec (3.1.5):

"An acknowledgment of an alarm does not cause a single alarm event to be acknowledged but all alarm events for the specific object with the associated alarm code id. This approach simplifies both in implementation but also in handling - if many alarms occur on the same equipment with short time intervals.

A suspend of an alarm causes all alarms from the specific object with the associated alarm code id to be suspended. This means that alarm messages stops being sent from the site as long as the suspension is active. As soon as the suspension is inactivated alarms can be sent again."

But when the simulators are used, this does not happen. If an alarm becomes active, it is sent to the supervisor simulator nonetheless even if it is suspended or if another alarm of the same component is suspended

otterdahl, 2023-05-17:
Thank for reporting this. Confirmed. This looks like a bug.

otterdahl, 2023-10-24:
Should be fixed with 1228abf5b5e964465292c322d783dfac71af58a8
It will be fixed with the upcoming 1.0.4 release

otterdahl, 2023-10-25:
Should be fixed with 1.0.4 release

closed 2023-10-25 by otterdahl