rsmp-nordic/rsmp_simulator #32: Simulator does not handle fP and fS correctly

daniyalyasin93, 2023-02-19:
I have been testing the simulator with the TLC SXL, as per my reading of the specification, and the comments by @otterdahl here: https://github.com/rsmp-nordic/rsmp_core/issues/45#issuecomment-619331369

My conclusion is that if I send the M0001 command with a value to the roadside equipment simulator and then when I send aggregated status from the road side equipment, it should have the fP set. But this doesn't happen, the simulator shows nothing in the fP and fS sections.

Is this a known bug? Do we plan to fix it at some time?

otterdahl, 2023-03-03:
I'd like to know more about this issue since I can't recreate it. If I try to send aggregated status with fP, it shows up in the list called "Aggregated status events" according to the screenshot below.

[image]

However, I do agree that is would be nice if the functional position/status could be highlighted in the FunctionalPosition/FuntionalState lists.

daniyalyasin93, 2023-03-21:
Hello

Thank you for the detailed response, I can confirm it works as you described. It appears that there is a small issue with the parser such that if the output is a csv generated from the sxl-tools repo, it doesn't parse all the enumeration values, but if the same file is saved using Microsoft Excel, it works

But I have a follow up question, in the simulator how do you use the "M0001" command to set this function state, it seems that the only way to set it is to just select the appropriate value in the road side simulator. Is there any way to set the current fP value from the scada simulator?

otterdahl, 2023-03-21:
We don't use the FunctionalPosition (fP) or FunctionalState (fS) in the SXL for TLC's. If you check the [aggregated status in the SXL](https://rsmp-nordic.org/rsmp_specifications/rsmp_sxl_traffic_lights/1.1/sxl_traffic_light_controller.html#aggregated-status) you can see that those fields are left empty and the description text says: `functionalPosition and functionalState not used (set to null)`. However, FunctionalPosition (fP) or FunctionalState (fS) may be used in SXL's for other types of equipment.

If you have found them in an SXL (included with the simulator?) then this is an error.

otterdahl, 2023-10-12:
This issue should be fixed with recent versions. Feel free to open it again if there are issues remaining.

closed 2023-10-12 by otterdahl

emiltin, 2023-10-12:
hi @daniyalyasin93, you could also have a look at the RSMP gem, which is a Ruby library and CLI-tool. It includes an TLC simulator that handles all the TLC commands and statuses. You can for example use M0001 to change functional position, and then read it using status requests, or just view the log to check what changes:
https://github.com/rsmp-nordic/rsmp

daniyalyasin93, 2023-10-13:
I will check out the gem. Sorry for the late reply. Thank you for your support