rsmp-nordic/rsmp_simulator #66: Trojan:Win32/Wacatac.F!ml

mousavo, 2023-10-16:
[image]
[image]

Hello, when I downloaded these 2 simulators, I could not run them on my PC as they were blocked by Microsoft's antivirus program.
Does it sound familiar to you?
What is the solution?

otterdahl, 2023-10-20:
I believe this is a false positive. I can't replicate this on my own Windows 11 PC despite scanning for viruses.

Anyway, an alternative to the setup executables is to try downloading [the setup zip file instead](https://github.com/rsmp-nordic/rsmp_simulator/releases/download/1.0.3/Setup_1_0_3.zip). The zip file contains everything needed and the simulator can be run directly from any folder.

otterdahl, 2023-12-01:
Is this still an issue with the latest releases?